morrownr/8812au-20210629 #94: building 8812au on 6.2 kernel

sjtmbg, 2023-05-05:
After upgrading to ubuntu 23.04 (6.2.0 kernel) the source in 8812au-20210629-main would no longer compile for me.  I copied over 3 source files from the RinCat RTL88x2BU-Linux-Driver github project:

os_dep/osdep_service.c
os_dep/linux/ioctl_cfg80211.c
os_dep/linux/os_intfs.c

and used them in place of the similar 8812au files and the driver built successfully.  Don't know what problems I may come across from having done this, but for now I am getting good download speeds.

morrownr, 2023-05-05:
Hi @sjtmbg 

I am working this issue this morning.

I am currently changing my main dev box to Ubuntu 23.04 as my testing over the last 6 months makes me think Ubuntu is the best platform for what I do.

I just tested the driver with both a manual installation and using install-driver.sh and it was totally clean and working well. If you "upgraded" your Ubuntu install of doing a clean install, did you run remove-driver.sh before the upgrade?

I am going to compare the 3 files you mentioned above to see what the differences are.

Could you be more specific about what you saw when the driver would not compile?

How often do you update this driver on your system? It is under constant development and you should as a minimum update before any system upgrades, any kernels updates and every 3 months or so would probably be good if no system updates or kernel updates are on the table.

sjtmbg, 2023-05-05:
I haven't been using the install- and remove-driver scripts but rather just compiling by using make clean; make; make install.  I do this every time there is a kernel upgrade, maybe twice a month on ubuntu.  I do check github periodically for updates but maybe I'm not looking in the right place or interpreting what I'm seeing there right.  I'm very much of a github newbie.

Here are the compile errors I saw when compiling after the 23.04/6.2.0 upgrade (gcc 12.2.0):

.../8812au-20210629-main/os_dep/osdep_service.c:2918:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2918 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
cc1: some warnings being treated as errors

.../8812au-20210629-main/os_dep/linux/os_intfs.c:2157:9: error: too many arguments to function ‘netif_napi_add’
 2157 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from .../8812au-20210629-main/include/osdep_service_linux.h:31,
                 from .../8812au-20210629-main/include/osdep_service.h:54,
                 from .../8812au-20210629-main/include/drv_types.h:27,
                 from .../8812au-20210629-main/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2586:1: note: declared here
 2586 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:260: .../8812au-20210629-main/os_dep/linux/os_intfs.o] Error 1

.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:467:17: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
  467 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .../8812au-20210629-main/include/osdep_service_linux.h:98,
                 from .../8812au-20210629-main/include/osdep_service.h:54,
                 from .../8812au-20210629-main/include/drv_types.h:27,
                 from .../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8303:6: note: declared here
 8303 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:478:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  478 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8287:6: note: declared here
 8287 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:1142:26: error: ‘struct cfg80211_roam_info’ has no member named ‘bssid’
 1142 |                 roam_info.bssid = cur_network->network.MacAddress;
      |                          ^
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c: At top level:
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10374:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, struct key_params *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, struct key_params *)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, struct key_params *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, struct key_params *)’} [-Werror=incompatible-pointer-types]
10374 |         .add_key = cfg80211_rtw_add_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10374:20: note: (near initialization for ‘rtw_cfg80211_ops.add_key’)
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10375:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))’ {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))’} [-Werror=incompatible-pointer-types]
10375 |         .get_key = cfg80211_rtw_get_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10375:20: note: (near initialization for ‘rtw_cfg80211_ops.get_key’)
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10376:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *)’} [-Werror=incompatible-pointer-types]
10376 |         .del_key = cfg80211_rtw_del_key,
[... cut ...]
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10459:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
.../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.c:10695:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10695 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:260: .../8812au-20210629-main/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:2026: .../8812au-20210629-main] Error 2

morrownr, 2023-05-05:
Okay. I understand and I promise I won't give a long lecture about github followed by a test. I'm not an expert either. I just learn what I need as I go along.

If you are using Ubuntu and you have an adapter based on the 8812au chipset, you can save yourself some effort if you so desire.

If you make sure everything is installed for Ubuntu per installation step 3, which is:

```
sudo apt install -y build-essential dkms git iw
```

...then, here is my recommendation:

```
sudo sh remove-driver.sh
```

Delete the directory you are using for the driver,

Following installation steps 3-8 so that you have the latest code. The problems you were having were because you have not been updating the driver code to match this repo. Development with Linux is continuous and you have to be able to stay current or things will blow with these out-of-kernel drivers. Here is how you do that:

Once you have fully installed per the instructions and rebooted:

git pull
sudo sh install-driver.sh

Do that about once every 3 months or anytime you do a major upgrade such as to 23.10...before you start the upgrade. 

The install-driver.sh script checks to see if the driver is already installed and removes it before it compiles the updated version. dkms will take care of any compiles needed when minor kernel updates flow in. You don't have to do anything.

This is about as easy as I can make it.

FYI: The main menu to this site is:

https://github.com/morrownr/USB-WiFi

Let me know how it goes.

@morrownr

sjtmbg, 2023-05-05:
Thanks for sparing me the lecture, lol - in return I promise to learn at least a wee bit about github!

I did the git clone and, just as you said, the build went flawlessly.

Just so I can better understand what I was doing wrong - when I download the zip file from the green "Code" drop-down menu, I'm not getting the most current version of what's in github - is that correct?

morrownr, 2023-05-08:
> Just so I can better understand what I was doing wrong - when I download the zip file from the green "Code" drop-down menu, I'm not getting the most current version of what's in github - is that correct?

Downloading the zip file should give you the most current code in the repo. I recommend you use clone instead of zip files. I've tested at length and the installations steps I have posted provide the most trouble free solution.

Remember what I said about near continuous changes to the code in the repo. It is not easy for a user to identify changes that that require them to update the code on their system. With this in mind, I have worked it down into 2 lines that users should run to update the code and driver on their system:

git pull
sudo sh install-driver.sh

You can run those 2 lines anytime you want. You could run them 5 times today without any problems as install-driver.sh recognizes previous installations and removes them before it compiles the code.

The truth is that out-of-kernel drivers are not the best solution for Linux given it dev model. Linux, in an effort to take advantage of the resources that are available, uses a dev model that that takes small bites instead of one big bite. Windows goes through a dev cycle that can take years and then the API's are locked and more or less stay locked for several years. The Linux dev cycle is 2.5 months. The standard for Linux is that drivers should be part of the Linux kernel. This is what I call in-kernel drivers. In this way, the devs are aware of changes and adjust accordingly. Their work also undergoes quality control and from a user perspective, there is nothing to do but plug the adapter in and it works.

One of the most popular menu items at the Main Menu here is:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

That document is very popular as it helps Linux users identify usb wifi adapters that use in-kernel drivers. The document provide many reviews and links to specific products. Did you know there are tri-band usb wifi adapters that are plug and play on Linux? Take a look at the above link and let me know if you have questions.

@morrownr

morrownr, 2023-05-11:
I'm going to mark this issue as solved but am going to leave it active as it contains information others can benefit from.

@morrownr