morrownr/8812au-20210629 #95: Wi-Fi option not showing up after install (Kernel 6.2.6)

fortyone041, 2023-05-08:
Distro : Pop!_OS 22.04 LTS
Kernel : 6.2.6-76060206-generic

The driver was installed successfully using `sudo ./install-driver.sh` with no errors but Wi-Fi option is not showing up. I have done `sudo ./remove-driver.sh` and reinstalled the driver again but nothing changed. Device ID is `2357:0122` which is in `supported-device-IDs` list

```
**lsusb**
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 003: ID 2357:0122 TP-Link 802.11n NIC
Bus 001 Device 002: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 005: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**rfkill list all**
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

**dkms status**
nvidia/525.105.17, 6.2.6-76060206-generic, x86_64: installed
rtl8812au/5.13.6, 6.2.6-76060206-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.2.6-76060206-generic, x86_64: installed
system76_acpi/1.0.2~1678132042~22.04~ed01124, 6.2.6-76060206-generic, x86_64: installed (original_module exists)
system76-io/1.0.2~1655490480~22.04~0217576, 6.2.6-76060206-generic, x86_64: installed

**iw reg get**
global
country MY: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 24), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5650 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 24), (N/A)

**iw dev gives me nothing**
```
Please let me know if you need anymore information from me. Thanks!

morrownr, 2023-05-10:
The VID/PID needed to be added to the driver.

Try this from the driver directory:

$ git pull
$ sudo sh install-driver.sh

See if that does it. The first line with pull the update and the second will install that driver. Don't worry about not removing the installed driver as the script will detect and remove it for you.

morrownr, 2023-05-11:
@fortyone041 

I'm sure you saw the messages from other users in another section of this repo. The best evidence I have right now is that the ID 2357:0122 VID/PID for your adapter should indeed be in this driver and I have restored it. I am adding this extra message as a reminder to me in the future.

There does seem to be some evidence that TP-Link also used this VID/PID in some rtl8812bu based adapters and that is a BIG NO NO.

Hopefully things comtinue to work for you.

@morrownr