morrownr/8812au-20210629 #98: Add support for TP-Link Archer T2U Plus

dlmyb, 2023-05-29:
```
$ lsusb 
Bus 002 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8812AU]
```

Spec is here [https://www.tp-link.com/us/home-networking/usb-adapter/archer-t2u-plus](https://www.tp-link.com/us/home-networking/usb-adapter/archer-t2u-plus).

I tried myself and added the PID/VID to `usb_intf.c`, it works.

morrownr, 2023-05-31:
Hi @dlmyb 

The link you provided says the adapter is a AC600 class adapter. The rtl8812au chipset is not a AC600 class chipset. It is a AC1200 class chipset.

The driver you should be using, if the info you posted is correct, is:

https://github.com/morrownr/8821au-20210708

The VID/PID is already in that driver.

dlmyb, 2023-05-31:
@morrownr Thanks for your correction. In fact later I found the dongle didn't work with 8812au driver, despite `iw list` output something.