morrownr/8812au-20210629 #101: exports duplicate symbol

talking-toaster, 2023-06-20:
[  312.494266] usb 1-2: new high-speed USB device number 5 using tegra-xusb
[  312.647187] usb 1-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  312.647196] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  312.647200] usb 1-2: Product: 802.11n NIC
[  312.647203] usb 1-2: Manufacturer: Realtek
[  312.647206] usb 1-2: SerialNumber: 123456
[  312.691809] cfg80211: exports duplicate symbol __cfg80211_alloc_event_net_skb (owned by kernel)

talking-toaster, 2023-06-20:
device:nvidia orin nx
os: ubuntu20.04
```
nvidia@nvidia-desktop:~$ uname -r
5.10.104-tegra
nvidia@nvidia-desktop:~$ uname -a
Linux nvidia-desktop 5.10.104-tegra #32 SMP PREEMPT Tue Apr 11 17:54:52 CST 2023 aarch64 aarch64 aarch64 GNU/Linux
```

morrownr, 2023-06-20:
Hi @talking-toaster 

Interesting. However, is there a problem with the operation of the adapter?

morrownr, 2023-08-09:
Hi @talking-toaster 

Somehow I missed this issue back when it was first posted. Is this still a problem?

FYI: I've seen a couple of users experience problems when running Realtek out-of-kernel drivers and nvidia out-of-kernel drivers on the same system. I don't have any nvidia cards so have no good way to try to duplicate the problem.