morrownr/8812au-20210629 #102: Kali 2023.2 6.3 kernel - interface not appearing after loading ko

bpinske, 2023-07-15:
I have installed the latest version here, however, I do not appear to see the interface actually visible for use. I expect to see wlan2 to appear. wlan0 and wlan1 are other unrelated wireless NICs I have.

I will note I also needed to run the following command myself to load the ko. But it did load correctly. 
insmod /lib/modules/6.3.0-kali1-amd64/updates/dkms/8812au.ko

Unclear to me on how to proceed.

https://www.amazon.com/Network-AWUS036ACS-Wide-Coverage-Dual-Band-High-Sensitivity/dp/B0752CTSGD
```
└─# ip link 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether d8:5e:d3:0c:e1:35 brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DORMANT group default qlen 1000
    link/ether 2a:dd:34:54:11:7d brd ff:ff:ff:ff:ff:ff permaddr 44:af:28:61:51:ac
4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DORMANT group default qlen 1000
    link/ether 00:02:6f:cc:05:6a brd ff:ff:ff:ff:ff:ff
```

```
└─# uname -a
Linux kali 6.3.0-kali1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.3.7-1kali1 (2023-06-29) x86_64 GNU/Linux

└─# lsmod | grep 8812
8812au               2625536  0

└─# lsusb | grep 8812
Bus 001 Device 002: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]

[ 1826.096145] RTW: module init start
[ 1826.096146] RTW: rtl8812au v5.13.6-15-gc40b977e2.20210629
[ 1826.096185] usbcore: registered new interface driver rtl8812au
[ 1826.096185] RTW: module init ret=0
```

bpinske, 2023-07-15:
I have been misreading 8821AU as 8812AU for the past hour :(

Works fine if I actually do the right thing.

closed 2023-07-15 by bpinske

morrownr, 2023-07-16:
Hi @bpinske 

Glad you got it sorted out. I'm just now getting around to taking a look. The 8821au and 8812au drivers can be confusing because at one time, the 8812au driver also supported the 8821au.