morrownr/8812au-20210629 #105: Ubuntu 22.04 driver install issue for  kernel: 6.4.6-060406-generic (x86_64)

niroop893, 2023-08-14:
Building module:
cleaning build area...
./dkms-make.sh...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.13.6 not found
Error! Bad return status for module build on kernel: 6.4.6-060406-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.45 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2023-08-14:
Hi @niroop893 

The first thing to look at, in my opinion, would be gcc version. You may need to install a more recent version than you have. It is possible to install and use multiple versions of gcc. The following guide can help you do so:

https://github.com/morrownr/USB-WiFi

10. How to install multiple gcc compiler versions

You can check which version of gcc your system is using right now:

$ gcc --version

niroop893, 2023-08-24:
Let me check with the gcc version