morrownr/8812au-20210629 #107: No /lib/modules/<kernel version>/build directory on RaspiOS

fhteagle, 2023-09-02:
Attempting to follow the instructions at https://github.com/morrownr/8812au-20210629#readme

but stuck on the step run ```sudo ./install-drivers.sh``` . The script is complaining "Your kernel header files aren't properly installed.", probably because /lib/modules/6.1.21-v8+/build does not exist. Reverting to the manual portion of the instructions, where ```make``` fails immediately at ```make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop. ```  

/lib/modules/6.1.21+/build and /lib/modules/6.1.21-v7+/build directories do exist (as a symlink into /usr/src/<kernelversion> , but not for 6.1.21-v8+ .

Not having the best brainpower day, so if I am missing something small and silly, please let me know. Otherwise, I am guessing its just a fluke of the current revisions of the packages?!?

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```
``` 
$ uname -a
Linux <hostname_censored> 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```
``` 
$ sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git
...
bc is already the newest version (1.07.1-2+b1).
build-essential is already the newest version (12.9).
dkms is already the newest version (2.8.4-3).
git is already the newest version (1:2.30.2-1+deb11u2).
raspberrypi-kernel is already the newest version (1:1.20230405-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
``` 
$ sudo ls -ald /lib/modules/6.1.21-v8+/*
drwxr-xr-x 11 root root   4096 Apr  6 16:10 /lib/modules/6.1.21-v8+/kernel
-rw-r--r--  1 root root 617502 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.alias
-rw-r--r--  1 root root 644598 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.alias.bin
-rw-r--r--  1 root root  15735 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.builtin
-rw-r--r--  1 root root  29866 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.builtin.alias.bin
-rw-r--r--  1 root root  17558 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.builtin.bin
-rw-r--r--  1 root root  90568 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.builtin.modinfo
-rw-r--r--  1 root root 232115 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.dep
-rw-r--r--  1 root root 311525 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.dep.bin
-rw-r--r--  1 root root    384 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.devname
-rw-r--r--  1 root root  68995 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.order
-rw-r--r--  1 root root    949 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.softdep
-rw-r--r--  1 root root 290268 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.symbols
-rw-r--r--  1 root root 352511 Apr  5 05:32 /lib/modules/6.1.21-v8+/modules.symbols.bin
```

morrownr, 2023-09-02:
Hi @fhteagle 

Having bad day? It happens.

Here is a link to the FAQ.md that I am currently working on and the updated info will work its way intp the FAQ file here as soon as I have time:

https://github.com/morrownr/rtl8852bu/blob/1.19.3/FAQ.md

Go to the very last section of the file. That problem is probably what you are experiencing.

FYI: I can't fix that. The RasPi devs need to fix that.

Don't forget to run the following before you continue:

$ make clean

Let me know.

fhteagle, 2023-09-03:
That was exactly the problem. Adding the /boot/config.txt option of arm_64bit=0 put me back on 6.1.21-v7l+ kernel, which had the build directory. install script worked as expected.

As an aside, all the reference material for the Edimax AC600 (USB id 7392:a812) referred to rl8812au driver, which was not picking up the adapter. But you actually helped me twice today via this old issue: https://github.com/morrownr/8812au-20210629/issues/27  , so I compiled rtl8821au which did pick up the adapter. Hostapd testing it next........

So yeah, its been that kind of day, but fortunately your repos come through again. Thanks Nick.

closed 2023-09-03 by fhteagle

morrownr, 2023-09-03:
Glad I could help.

I use the 64 bit version of the RasPiOS on both of my Pi's. It does not suffer the same illness as the 32 bit version.

My experience is that going from RasPiOS to something else is like jumping from the frying pan into the fire. It doesn't get better, it gets worse. I'd like to see the RasPi devs hurry up and rebase on the new Debian 12 as that will update a lot of wifi related things.

fhteagle, 2023-09-03:
Yeah I thought I had installed the 64 bit variant of RaspiOS when I last installed it on that pi a few years ago. Apparently not. Oops.

I prefer the arch package management / AUR system by far. I run Endeavour on my laptop and home router/server (both x86_64), so I am plenty familiar with arch. Hopefully not too many wrinkles in the rug on their handling of the continuum of raspi quirks. Only one way to find out..............................

morrownr, 2023-09-03:
Of the time I spend helping users, I'll estimate around a third of the issues are RasPi users and the issue is not with the driver, it is with something broken in the RasPiOS. I'll go further, I'd estimate around 90% of the resported issues are because of a problem in the distro, the driver. Heck, it may be more than 90%.

Now you see why I recommend adapters with in-kernel drivers...

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Realtek's drivers are for programmers working on embedded projects, not for us. You can put lipstick on a pig but it is still a pig.