morrownr/8812au-20210629 #108: Error! Bad return status for module build on kernel: 6.1.46-08797-g9c18484153bc (aarch64)

cray12399, 2023-09-03:
Hello, I am trying to install RTL8812AU on my Libre Renegade 4GB board. It has Raspberry Pi OS Lite 64bit installed on it and is running kernel 6.1.46-08797-g9c18484153bc. The distro and packages are fully upgraded according to `apt update/upgrade/full-upgrade/dist-upgrade`. I followed all of the instructions on the README perfectly, yet I am getting the following errors:

When I trying to install via the script, this is what I get:

```
: ---------------------------
: install-driver.sh v20230628
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 4049408 (total system memory)
: 6.1.46-08797-g9c18484153bc (kernel version)
: gcc (Debian 10.2.1-6) 10.2.1 20210110
: dkms:2.8
: mokutil not installed
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6

Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source ->
                 /usr/src/rtl8812au-5.13.6

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh..................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.46-08797-g9c18484153bc (aarch64)
Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

When I try to make the installation manually:

```
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.46-08797-g9c18484153bc/build M=/home/pi-ap/src/8812au-20210629  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.46-08797-g9c18484153bc'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /home/pi-ap/src/8812au-20210629/core/rtw_cmd.o
scripts/basic/fixdep: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by scripts/basic/fixdep)
scripts/basic/fixdep: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by scripts/basic/fixdep)
make[2]: *** [scripts/Makefile.build:250: /home/pi-ap/src/8812au-20210629/core/rtw_cmd.o] Error 1
make[2]: *** Deleting file '/home/pi-ap/src/8812au-20210629/core/rtw_cmd.o'
make[1]: *** [Makefile:2012: /home/pi-ap/src/8812au-20210629] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.46-08797-g9c18484153bc'
make: *** [Makefile:2504: modules] Error 2
```

What do you advise that I try to do in fixing or further diagnosing this problem?

morrownr, 2023-09-03:
Hi @cray12399 

> What do you advise that I try to do in fixing or further diagnosing this problem?

Send me a link for the hardware: Libre Renegade 4GB board

You might try installing gcc12. I have a guide to help but you only need to install the extras you want:

https://github.com/morrownr/USB-WiFi

Go to menu item 10.

cray12399, 2023-09-03:
@morrownr Thank you for responding. I actually got it working. The problem is that my Glibc version was 2.31. However, according to the stock repos, I was up to date. So the solution I found was the add the add the "sid" repo to my sources list. Here are the steps I took to get it working for anyone else who has the same issue:

1. Add the following line to the file /etc/apt/sources.list: `deb http://ftp.debian.org/debian experimental main`
2. `sudo apt update`
3. 'apt-get -t sid install libc6-amd64 libc6-dev libc6-dbg'
4. Reboot
5. Comment out the above line in sources.list so that it doesnt try to update your repo.

After that, I just followed the steps on the README and the install went without a hitch.

Thank you for the driver @morrownr!

closed 2023-09-03 by cray12399

morrownr, 2023-09-03:
> The problem is that my Glibc version was 2.31.

Glad you got it going. I saw the glibc issue and it was on my list of things to do but you have a very old base distro teamed up with a pretty modern kernel which can and will cause problems.  Going more than one LTS beyond the kernel that shipped with a distro release starts causing problems and that happened in this case. Sometimes you just need to save the important files and do a fresh install with the most current version.