morrownr/8812au-20210629 #110: Installed, but interface not showing up Debian 12 Kernel 6.4

KhanSW, 2023-10-04:
So everything is installed correctly but the interface is not showing up as a device that can be used.

**dkms status**

`rtl8812au/5.13.6, 6.4.0-0.deb12.2-amd64, x86_64: installed`

**lsusb**

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 001 Device 004: ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader

Bus 001 Device 003: ID 13d3:56a6 IMC Networks Integrated Camera

Bus 001 Device 002: ID 8087:0032 Intel Corp. AX210 Bluetooth

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**ip a**

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever

2: enp0s31f6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether e8:6a:64:9c:26:a0 brd ff:ff:ff:ff:ff:ff

3: wlp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 64:49:7d:e7:78:b8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.150/24 brd 192.168.1.255 scope global dynamic noprefixroute wlp3s0
       valid_lft 85299sec preferred_lft 85299sec
    inet6 2604:2d80:9788:4c00:6978:1d1a:e591:d955/64 scope global dynamic noprefixroute 
       valid_lft 1002858sec preferred_lft 398058sec
    inet6 fe80::518a:bd7f:d98d:d432/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

Is there something I'm missing? I uninstalled using the script prior to upgrading the kernel to 6.4 ( as I was having the same issue in the 6.1 kernel and had hoped by upgrading it would fix my problem). After upgrading I attempted to update the src but it was already up to date, and ran the install shell script again which re-installed the driver.) No matter what I do I can not get the interface to show up, only my internal Intel AX210 wifi card shows up. I have tried switching between usb2 to usb3 in the conf file, but that has not helped either. Hoping someone can nudge me in the right direction here.

KhanSW, 2023-10-04:
Weird, it would not work properly using the dkms install so I decided to try and compile it from source using make; make install and that is working correctly. 

Any reason why the dkms method did not work correctly?

KhanSW, 2023-10-04:
I used the install-driver.sh script.

KhanSW, 2023-10-04:
Just to test it out, I rolled over to Linux Mint 21 and attempted the install there. Using the install.sh script worked perfectly on that distro and I had no issues. Up and running in less than three minutes. I can roll back if you would like to test it further later tonight. I have to take my motorcycle to shop to get serviced and will be gone for 5+ hours. I would be more than happy to test if further.

morrownr, 2023-10-04:
I am going to be out of the office starting in a few minutes for 5+ hours so no big rush.

I had a somewhat similar bug report on another rtl driver about a week ago and I added a little code to install-driver.sh. The user reported his problem as fixed and that same fix may be what is needed here so I'll get on that and merge sometime today if all goes well.

What kind of bike do you have? Mine is a Honda Rebel 500.

morrownr, 2023-10-05:
@KhanSW @sollted 

I have merged my best bet as what I think the problem may be. If you can roll back, if necessary, and then test, that would be appreciated. You can update your driver by running:

$ git pull
$ sudo sh install-driver.sh

Also, of note, the installation script has code to detect and remove previous installations so you really do not need to uninstall previous drivers before running the installation script.

For what it is worth, my main dev box is running LMDE 6 which is Debian 12 under the Cinnamon desktop. I installed this driver, it worked. I think problem only comes up in certain situations where a different version of this driver from other repos has been installed at some point.

KhanSW, 2023-10-05:
Ok giving it a shot right now, let you know how it went in a few minutes.

KhanSW, 2023-10-05:
ok so fresh install of Debian 12, updated & updated and followed instructions exactly as last time. Same status, dkms shows installed but when I plug in the adapter it is still not activating.

sudo dkms status:
rtl8812au/5.13.6, 6.1.0-12-amd64, x86_64: installed

lsusb does not show adapter but every other distro I install it in works properly  following your instructions. This is only happening in Debian 12. I believe it must be missing something. Now yesterday when I compiled it from source it did work, just not with the DKMS install script. So before I compile from source, is there anything else you want to test, or anything I can provide from my end?

KhanSW, 2023-10-05:
Oh forgot to add, I ride a 2021 Harley Davidson Road King Special

KhanSW, 2023-10-05:
On my Debian 12 install I am running Gnome.

morrownr, 2023-10-06:
Test complete.

I did a clean install Debian 12.1 with Gnome to a partition on an x86_64 desktop system. I cloned the driver and installed per the instruction steps in the README. I was able to log in and use the system online.

So much for finding a bug. What is the difference between my Debian 12.1 setup and yours. All ideas welcome. Without being able to duplicate the problem here, this may be a challenge.

KhanSW, 2023-10-09:
So I completely wiped my system, did a fresh install of Debian 12 (not Linux Mint-DE) as I suspect Mint adds Realtek Firmware by default (Can you check under additional drivers and see if they are listed) as I suspect this may be whats missing.

So clean install using script and I am getting  the correct status under dkms, and it does appear under lsusb. Here is the following outputs:

khansw@T480-Debian:~$ sudo dkms status
**rtl8812au/5.13.6, 6.1.0-13-amd64, x86_64: installed**

khansw@T480-Debian:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter**
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader
Bus 001 Device 003: ID 13d3:56a6 IMC Networks Integrated Camera
Bus 001 Device 002: ID 8087:0032 Intel Corp. AX210 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

However the adapter is not selectable. For example when I check iwconfig it only lists my internal AX210 Intel Wifi card.

khansw@T480-Debian:~$ sudo iwconfig
lo        no wireless extensions.

enp0s31f6  no wireless extensions.

wlp3s0    IEEE 802.11  ESSID:"REDACTED"  
          Mode:Managed  Frequency:5.805 GHz  Access Point: REDACTED   
          Bit Rate=780 Mb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=63/70  Signal level=-47 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:387   Missed beacon:0

So like I said Im wondering if its still missing something. On your test did you run it under Mint-DE or Debian 12 Stock?

morrownr, 2023-10-09:
> as I suspect Mint adds Realtek Firmware by default

There is no firmware to add for this driver. The firmware for this driver is included here in the driver.

> On your test did you run it under Mint-DE or Debian 12 Stock?

Debian 12 Stock with Gnome

> However the adapter is not selectable. For example when I check iwconfig it only lists my internal AX210 Intel Wifi card.

I wouldn't use iwconfig as it has long ago been depreciated, Use 'iw", you may have to install it (if you set up sudo).

$ sudo apt install iw
$ iw dev

I have no explanation at this point. I'll keep an open eye and post if I figure something out. Keep me posted.

sollted, 2023-10-10:
output of my sudo iw:
devphy#0
	Interface wlp3s0
		ifindex 4
		wdev 0x1
		addr 
		ssid 
		type managed
		channel 7 (2442 MHz), width: 20 MHz, center1: 2442 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

morrownr, 2023-10-10:
> Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

If your adapter is incorrectly marked as using a rtl8812au chipset with an ID of 0bda:8812, then you need a different driver.

What you could do is try the rtl8812bu chipset driver (88x2bu) but you would need to add a line for this ID. You need to remove this driver as in 'remove-driver.sh' and delete the directory and then go to:

https://github.com/morrownr/88x2bu-20210702/tree/main

The following file shows which file needs to the modification:

https://github.com/morrownr/88x2bu-20210702/blob/main/supported-device-IDs

If it is not clear what to do, let me know.

sollted, 2023-10-10:
no this is my other card the alfa AWUS035ACH is not showing

morrownr, 2023-10-10:
@sollted 

I have a Alfa ACH. That is what I was test with. It definitely uses a rtl8812au chipset. Let me crosscheck again.