morrownr/8812au-20210629 #114: Need testers and coders: I have am updated version of the code for the rtl8812au chipset

morrownr, 2023-12-21:
The new code is dated 20210820. The information I was given is that this newer code is the final release of the driver for the rtl8812au chipset. My first thought was to merge the changes in the driver in this repo as the new version is only about 2 months newer than this version but when I compared the code, it was obvious that a lot of code had been added and changed so I think it is best to start clean and carefully get this version up to speed before making it public.

I am requesting that anyone interested in testing or working on the code reply to this message and I will add you to the new repo. The new repo is private for now. The amount of time this project will take could be as little as 2 months but could be much longer. We need to do a good job as this will likely be the go-to driver for the rtl8812au chipset for many years for Linux users around the world.

Looking forward to hearing from you.

@morrownr

castillofrancodamian, 2023-12-21:
I want to try it.

morrownr, 2023-12-22:
Hi @castillofrancodamian 

You have been added to the repo. The repo is private while we get it ready which may take some time. That means you have to log into github to be able to see the repo.

The repo is:

git clone https://github.com/morrownr/8812au-20210820.git

The repo currently only contains the base driver as it comes from Realtek. It currently only supports up to kernel 5.11. I will start adding support for later kernels as I have time. What kernel are you using?

The installation and removal scripts will also be added as I have time so the only way to install and remove the driver until I do that is:

$ make clean
$ make
$ sudo make install

$ sudo make uninstall

FYI: I used to live in BA.

@morrownr

castillofrancodamian, 2023-12-22:
@morrownr 
```
[franco@pc ~]$ uname -a
Linux pc 6.6.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 14 Dec 2023 03:45:42 +0000 x86_64 GNU/Linux
```

morrownr, 2023-12-22:
It will take me some time to get to kernel 6.6 as this is tedious work. You are wasting your time trying to compile until you see kernel 6.5 support go in.

Remember that once you have cloned the repo, you can update it with the following command:

$ git pull

bungh0l10, 2024-01-03:
I'd also like to give it a try. Unfortunately on raspbian bookworm 6.1 means it won't work yet, right?

Cheers and thanks for your efforts.

0-xeno-0, 2024-01-03:
┌──(root㉿kali)-[/opt/8812au-20210629]
└─# sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 2/2 (in-use/total processing units)
: 2012064 (total system memory)
: 6.5.0-kali3-amd64 (kernel version)
: gcc-13 (version of gcc used to compile the kernel)
: gcc (Debian 13.2.0-5) 13.2.0
: dkms-3.0.12
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6
Error! Could not locate dkms.conf file.
File: /usr/src/rtl8812au-5.13.6/dkms.conf does not exist.
An error occurred. dkms add error:  4
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

Although I might not be a expert but I would love to help, and learn.

morrownr, 2024-01-03:
@castillofrancodamian @bungh0l10 @0-xeno-0 

All of you have been added to the repo. The repo is private for testing which means you have to be logged in to see it or clone it.

Here is the repo location:

https://github.com/morrownr/8812au-20210820

The current status is available in Issue 1. I am posting a log so that you can see what I do. I have a lot of work left but the driver should be testable. Please follow the installation steps in the README or do a manual installation or, preferably, test both but let me know which you are doing when you report results. If using `install-driver.sh` , please post the results of the on screen log. If do a manual installation, please post whatever shows up on the screen.

I need help with the docs. I need the docs, such as README and FAQ to be in clear international English. If you see anything that is not clear or does not translate well, Please let me know or submit a PR to make the change. There are guides available for you to learn how to do PRs so that you can directly make the changes and submit them.

I am requesting that you update the driver at least once per week and test. If you find a problem, post an issue. One issue per problem.

I do not know how long testing will take as I would like for this release to be a very solid release as it is based on the last source release from Realtek that we will see. My opinion is that this is the best out-of-kernel driver Realtek has ever released and given that many Linux users have adapters based on the rtl8812au chipset, hopefully we can keep this driver going for many years.

I'll see you at the new repo. Please check in by leaving a `hi, I'm here` message in Issue 2.

@morrownr

morrownr, 2024-01-03:
For got to mention:

An easy way to pull in an updates and compile is:

$ git pull
$ sudo sh install-driver.sh

Note: `install-driver.sh` contains code to detect and remove any previously installed driver for the 8812au so you do not need to uninstall before running `install-driver.sh`. If you do need to do a full installation to include deleting the driver directory, run:

sudo sh remove-driver.sh

This script will fully remove the driver and any additions that have been made and you can simply delete the driver directory.

frsarieg, 2024-01-06:
Please, I want to test it too. Thanks :-)

morrownr, 2024-01-06:
@frsarieg 

You have been added.

michelwi, 2024-01-10:
I am happy to help with translation and testing, if / once my kernel (6.1) is supported.

rojer, 2024-01-13:
please add me too.

morrownr, 2024-01-14:
@michelwi 

> I am happy to help with translation and testing, if / once my kernel (6.1) is supported.

Great. Kernels 4.19 to 6.7 are supported.

I need help with the docs. The docs are in English but what I need is work to ensure it is International English and will translate well on electronic translators. I am fluent in English and Spanish. The docs are contained in README,md, FAQ.md and 8812au.conf. I also need help to reduce the size of the docs so if you see ways to reduce the size while keeps the important info, please work that..

Latest, this project has no deadline as the version is up is pretty good. I would like to finish and make the new driver available at some point in the first half of this year.

Regards,

@morrownr

briantbutton, 2024-01-14:
How interesting!  Count me in.  I would like to test it.

P.S. I am using a WiFi dongle with the rtl8812au chipset.  It seems like I might be getting intermittent disconnects.  Hard to prove; hard to be sure.  It seems to happen when my attention is  elsewhere.

morrownr, 2024-01-15:
> How interesting! Count me in. I would like to test it.

Expect an invitation in email.

gbburkhardt, 2024-01-18:
"Works for me®" with Ubuntu 22.01, Netis WF2190, Linux version 6.5.0-14-generic.
Thanks so much for maintaining this driver.  It's a life saver!

dsvf, 2024-01-19:
I'm interested in testing

Ph0rk0z, 2024-01-22:
I'm on arch and using AWUS036ACH for scans. Currently on https://github.com/aircrack-ng/rtl8812au but when I scan with that it seems like it's not doing much better than my built in laptop radio in terms of 5ghz. 

This is a newer driver that supports monitor mode? 

Perhaps it could give me better scans? Someone had posted a long time ago about turning off agc https://github.com/aircrack-ng/rtl8812au/issues/242 and didn't get anything back. I'm not really interested in far reaching power here but I think with the 5dbi antennas, this should be doing better, no?

Since you maintain many drivers with these radios, thoughts?

michelwi, 2024-01-22:
> This is a newer driver that supports monitor mode?

The new driver currently does not support monitor mode. But the old driver (i.e. this one) should. Check the FAQ in this repo. Additionally https://github.com/morrownr/Monitor_Mode might be helpful.

Ph0rk0z, 2024-01-23:
I will try the current one as well and see if there is a difference. An AGC ramping up and down does sound like something that would really suck when driving around and surveying though.

Mauzy0x, 2024-01-31:
I would be interested in helping in anyway I can.

morrownr, 2024-02-03:
@Ph0rk0z @Mauzy0x 

I have added both of you to the repo. Expect an email with instructions.

Please read issue 1 first and then leave a message in issue 2. You help is appreciated.

We are in no hurry as the existing driver is pretty good. I'm look at a proposed release date of June 30.

Regards,

@morrownr

Ph0rk0z, 2024-02-03:
Thanks. Going to try it out and see how the scans go.

TrnHuPhc2909, 2024-02-07:
I'm interested with this driver. 
Currently having a TP-Link T4U v5 that came with rtl8812au and kernel 6.5.0 So it should work with this,
But I want to try it out if this driver can make T4U can use at UNII-3 band (TP-Link limit the signal and remove UNII-3) on their chipset.
Thanks

morrownr, 2024-02-12:
@TrnHuPhc2909 

> I'm interested with this driver.

I have added you to the repo, You will get an email with instructions. Please read issue 1 and then post a short msg in issue 2 so that I know that you are there. I hope you enjoy testing and have time to contribute.

@morrownr

morrownr, 2024-02-25:
@Dan-Milne 

Expect an email with instructions.