morrownr/8812au-20210629 #115: Multiple UBSAN: array-index-out-of-bounds errors.

mbs0221, 2023-12-25:
Mutiple array-index-out-of-bounds errors occurred under Linux 6.5.0.
```
[    4.842124] 8812au: loading out-of-tree module taints kernel.
[    4.842149] 8812au: module verification failed: signature and/or required key missing - tainting kernel
[    4.934904] RTW: rtl8812au v5.13.6-15-gc40b977e2.20210629
[    5.690350] usbcore: registered new interface driver rtl8812au
[    5.693788] rtl8812au 5-2:1.0 wlx502b73600db5: renamed from wlan0
[   13.714347] UBSAN: array-index-out-of-bounds in /home/public/8812au-20210629/core/rtw_wlan_util.c:1857:48
[   13.714415]  HT_caps_handler+0x58/0x4f0 [8812au]
[   13.714533]  OnAssocRsp+0x2d0/0x370 [8812au]
[   13.714641]  _mgt_dispatcher+0x73/0xb0 [8812au]
[   13.714732]  mgt_dispatcher+0x247/0x2b0 [8812au]
[   13.714823]  validate_recv_mgnt_frame+0x442/0x500 [8812au]
[   13.714944]  validate_recv_frame+0x200/0x450 [8812au]
[   13.715049]  recv_func_prehandle+0x19/0x40 [8812au]
[   13.715148]  recv_func+0x2c/0x150 [8812au]
[   13.715245]  rtw_recv_entry+0x19/0x30 [8812au]
[   13.715341]  pre_recv_entry+0x6c/0x140 [8812au]
[   13.715439]  recvbuf2recvframe+0x13c/0x2a0 [8812au]
[   13.715577]  usb_recv_tasklet+0x61/0x150 [8812au]
[   13.715842] UBSAN: array-index-out-of-bounds in /home/public/8812au-20210629/core/rtw_wlan_util.c:1862:75
[   13.715876]  HT_caps_handler+0xcd/0x4f0 [8812au]
[   13.715990]  OnAssocRsp+0x2d0/0x370 [8812au]
[   13.716090]  _mgt_dispatcher+0x73/0xb0 [8812au]
[   13.716183]  mgt_dispatcher+0x247/0x2b0 [8812au]
[   13.716274]  validate_recv_mgnt_frame+0x442/0x500 [8812au]
[   13.716393]  validate_recv_frame+0x200/0x450 [8812au]
[   13.716498]  recv_func_prehandle+0x19/0x40 [8812au]
[   13.716598]  recv_func+0x2c/0x150 [8812au]
[   13.716701]  rtw_recv_entry+0x19/0x30 [8812au]
[   13.716801]  pre_recv_entry+0x6c/0x140 [8812au]
[   13.716899]  recvbuf2recvframe+0x13c/0x2a0 [8812au]
[   13.717034]  usb_recv_tasklet+0x61/0x150 [8812au]
[   13.717295] UBSAN: array-index-out-of-bounds in /home/public/8812au-20210629/core/rtw_wlan_util.c:1868:76
[   13.717327]  HT_caps_handler+0x10d/0x4f0 [8812au]
[   13.717441]  OnAssocRsp+0x2d0/0x370 [8812au]
[   13.717541]  _mgt_dispatcher+0x73/0xb0 [8812au]
[   13.717637]  mgt_dispatcher+0x247/0x2b0 [8812au]
[   13.717728]  validate_recv_mgnt_frame+0x442/0x500 [8812au]
[   13.717847]  validate_recv_frame+0x200/0x450 [8812au]
[   13.717952]  recv_func_prehandle+0x19/0x40 [8812au]
[   13.718051]  recv_func+0x2c/0x150 [8812au]
[   13.718149]  rtw_recv_entry+0x19/0x30 [8812au]
[   13.718246]  pre_recv_entry+0x6c/0x140 [8812au]
[   13.718344]  recvbuf2recvframe+0x13c/0x2a0 [8812au]
[   13.718477]  usb_recv_tasklet+0x61/0x150 [8812au]
```

morrownr, 2023-12-25:
Thanks for the report. I've added this to my to-do list.

Is the driver working?

mbs0221, 2023-12-28:
> Thanks for the report. I've added this to my to-do list.
> 
> Is the driver working?

The driver is working.

morrownr, 2024-02-19:
@mbs0221 

I'm getting back to this issue. I am actually working on bringing a repo with a newer version of the code online soon. Myself and a few testers are working on it in a private repo. I have this issue on my to-do list but I am not seeing the issue.

Would you like access to the private repo so that you can do some bug hunting?

@morrownr

BecauseTheWorldIsRound, 2024-02-21:
In case it matters, same problem here, kernel 6.5.0-18, Realtek RTL8811AU

```
UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_wlan_util.c:1869:34
index 2 is out of range for type 'u8 [1]'
```
This is for the `8821au-20210708` repo.
Does the driver work? Somtimes it works intermittently, then I saw this error in `kern.log`.

Note that I had the same intermittent problem with Realtek  RTL88x2bu using the `88x2bu-20210702` repo.

(I own both, and switched to the 8811 due to the problems. I thought it was my setup -- could be anyway)

Thanks for taking care of that.

morrownr, 2024-02-21:
Hi @BecauseTheWorldIsRound 

Thanks for the info.

> Note that I had the same intermittent problem with Realtek RTL88x2bu using the 88x2bu-20210702 repo.

Have you tried the rtw88 in-kernel driver? The 6.5 kernel should have the in-kernel driver available. You would just need to run `sudo sh remove-driver.sh` to remove the `88x2bu-20210702` and then reboot.

morrownr, 2024-02-21:
> Actually I blacklisted the stock driver, does it cover both 8821 and 88x2bu?

You don't need to blacklist anything with the drivers here as the install-driver.sh script and remove-driver.sh scripts do that for you. Look in the files 88x2bu.conf and 8821cu.conf files and you will see the blacklist that is appropriate based on what is in the kernel. The scripts install and remove those files as appropriate.

BecauseTheWorldIsRound, 2024-02-22:
> You don't need to blacklist anything

When things don't go as expected, we tend to try various solutions :-)
Good news: the stock driver seem to be working.

By the way, the current problem might be linked to the kernel 6.5. I used your driver for 88x2bu a while ago without problems (was 5.x at the time iirc). It was on another box which config is not available anymore.

Anyway, I'll probably switch to a Mediatek based device, which seems to be more Linux/Openwrt friendly.

Thanks for your help.

dubhater, 2024-02-22:
If you change ``u8  data[1];`` to ``u8  data[];`` the warning probably goes away.

morrownr, 2024-03-15:
FYI: I finally had time to take a look at this. I'll try to merge a patch later today.

morrownr, 2024-03-16:
Patch merged. Give it a try and report.

$ git pull
$ sudo sh install-driver.sh

A new version of this driver should be released by the end of this month.

@morrownr