morrownr/8812au-20210629 #117: The driver doesn't build on Arch with 6.7 kernel

Finoderi, 2024-01-16:
Apparently something has changed and dkms cannot build the module against 6.7.arch3-1 kernel. But everything looks fine on 6.6.9.arch1-1.

morrownr, 2024-01-16:
Finding out what the problem is requires information: Can you run `install-driver.sh` and post the onscreen output?

Thanks,

@morrownr

getong, 2024-01-16:
```
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 88/88 (in-use/total processing units)
: 131765328 (total system memory)
: 6.7.0-arch3-1 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 13.2.1 20230801
: dkms-3.0.12
: This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
--> rtl8812au 5.13.6: added
Error! The module/version combo: rtl8812au-5.13.6: is not located in the DKMS tree.
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6
Error! DKMS tree already contains: rtl8812au-5.13.6
You cannot add the same module/version combo more than once.
This driver may already be installed.
Run the following and then reattempt installation.
$ sudo ./remove-driver.sh
```

morrownr, 2024-01-16:
> Error! The module/version combo: rtl8812au-5.13.6: is not located in the DKMS tree.

I've seen this twice lately on Arch based distros. I have code in the `install-driver.sh` script to seek out and delete previously installed rtl8812au drivers. It works well but on some distros, at times, it can fail if upgrading major versions of the kernel. Minor version upgrades are no problem. In this case you were going from 6.6 to 6.7 which is a major upgrade. A minor upgrade would be like 6.6.10 to 6.6.11 and that is never a problem. 

I warn about this problem in the README and give the following instructions:

```
Warning: If you decide to do a distro upgrade, which will likely install a new version of kernel such as 5.15 to 6.1, you need to update this driver with the newest available code and then run the removal script before performing the disto upgrade. Use the following commands in the driver directory:

git pull

Note: Do not reboot before running the below command so that the driver stays active until after your distro upgrade is complete.

sudo sh remove-driver.sh

Note: The following command will reinstall the updated driver after you are finished with the distro upgrade and reboot.

sudo sh install-driver.sh

```

To fix this problem:

sudo dkms remove rtl8812au/5.13.6 --all

sudo sh install-driver.sh

Let me know how it goes. I will take a look at the code in my scripts to see if there is something I have done wrong.

@morrownr

Finoderi, 2024-01-17:
OK. As far as I can tell the remove-driver.sh script doesn't work correctly every time.
After I launched it, rebooted, as it recommended, and launched install-driver.sh, I got the same output as @getong.
Then I looked into output of install-driver.sh:
` : remove-driver.sh v20231118
: x86_64 (kernel architecture)
: 6.7.0-arch3-1 (kernel version)
: ---------------------------

--> rtl8812au 5.13.6: added
Error! The module/version combo: rtl8812au-5.13.6: is not located in the DKMS tree.
Removing 8812au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8812au-5.13.6
The driver was removed successfully.
You may now delete the driver directory if desired.`

It says the driver is not in dkms tree. But:
`$ sudo dkms status
rtl8812au/5.13.6: added`

And if I manually remove it:
`$ sudo dkms remove rtl8812au/5.13.6 --all
Deleting module rtl8812au-5.13.6 completely from the DKMS tree.`
It does get removed and the driver can be successfully installed with install-driver.sh.

morrownr, 2024-01-17:
I appreciate both of you posting the info I needed to understand what the problem is. I will post a patch as soon as I am finished testing it. Hopefully it will end this problem.

@morrownr

morrownr, 2024-01-17:
Glad it is worked. I see a way to prevent this problem in the scripts so will work it as I have time.

@morrownr

john9631, 2024-02-23:
FYI.  I kept all 3 versions and used the same version to remove anything left over.  I do recall an issue once when I'd added but not built or installed.  Advice to check the status and use of

sudo dkms remove _rtl88x2bu/5.13.1_ --all

could also be added if you thought it would help (with the users driver and version substituted).

Just a note on terminals. I'm an xterm / tmux guy but downloaded warp terminal this week.  It was created for macs but has just been released for linux - free for non-commercial use - but its combination of a modern take on terminals + ai is interesting.  Has a good vim mode.

Finoderi, 2024-02-24:
Blurring the line between AI and human spam.

john9631, 2024-02-25:
Yes.   I was quite taken with aspects of it.  FWIW I will continue using xterm/tmux most of the time as it doesn't currently support sequential keystrokes like jk to replace escape in vim.  Also a small issue with ptpython although I could probably overcome that if the first was sorted out.