morrownr/8812au-20210629 #120: Request to add product ID for Netgear A6100 (ID 0846:9052) with RTL8812AU

qxray, 2024-02-21:
Hi,

checked the file "supported-device-IDs" and only 0846:9051 is included although 0846:9052 is also using RTL8812AU.

The link to the product: https://www.netgear.com/at/home/wifi/adapters/a6100/

Thanks and best regards,
Raimund

morrownr, 2024-02-21:
Hi @qxray 

This is not the correct driver for your adapter. Your adapter uses the rtl8821au chipset. The driver for the 8821au is:

https://github.com/morrownr/8821au-20210708

If you check "supported-device-IDs", you will see the vid/pid for your adapter.

@morrownr

qxray, 2024-02-21:
Hi,

Thanks a lot. Interestingly the 8812 driver allowed to scan and list the
available network ssids, but no connection. Will try with the 8821 driver.

Best regards
Raimund

morrownr, 2024-02-21:
> Interestingly the 8812 driver allowed to scan and list the
available network ssids, but no connection. Will try with the 8821 driver.

Hmmm...that should not happen so you likely have an old driver installed. If you have problems with the 8821au driver, you may need to investigate what you previously had installed.