morrownr/8812au-20210629 #125: 6.9.1 kernel fail

getong, 2024-05-17:
[make.log](https://github.com/morrownr/8812au-20210629/files/15358812/make.log)

```
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: 在函数‘rtw_cfg80211_ch_switch_notify’中:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:469:17: 错误：提供给函数‘cfg80211_ch_switch_started_notify’的实参太多
  469 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8811:6: 附注：在此声明
 8811 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/phydm/phydm_dfs.o
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:488:9: 错误：提供给函数‘cfg80211_ch_switch_notify’的实参太多
  488 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8795:6: 附注：在此声明
 8795 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
```

morrownr, 2024-05-18:
Hi @getong 

I need you to move on the new newer version of the driver:

https://github.com/morrownr/8812au-20210820

I have been busy and have failed to take this repo down.

In case you are interested, there is an effort to bring a standards compliant driver for the rtl8812au chip. I post when ready for testing.

getong, 2024-05-18:
Thanks, it works.

closed 2024-05-18 by getong