morrownr/8812au-20210629 #126: The driver fails to build on 6.9.1 kernel

Finoderi, 2024-05-18:
It's 6.9.1-arch1-1 kernel.
I tried to manually remove the driver, but building still fails. There was an error in build log about too many arguments. Sorry, i didn't copy it. If it's really needed I can try to install the driver again and produce that log.

nvzeppelin, 2024-05-19:
Same problem. Output:
`: ---------------------------
: install-driver.sh v20240117
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 16/16 (in-use/total processing units)
: 32785800 (total system memory)
: 6.9.1-arch1-1 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 14.1.1 20240507
: dkms-3.0.12
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
Module 8812au-5.6.4.2_35491.20191025 for kernel 6.9.1-arch1-1 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko.zst:
 - Uninstallation
   - Deleting from: /usr/lib/modules/6.9.1-arch1-1/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...
Deleting module 8812au-5.6.4.2_35491.20191025 completely from the DKMS tree.
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6
Creating symlink /var/lib/dkms/rtl8812au/5.13.6/source -> /usr/src/rtl8812au-5.13.6
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/modules/6.9.1-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
kernelver=6.9.1-arch1-1 ./dkms-make.sh.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.9.1-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.13.6/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh`
make.log:
`DKMS make.log for rtl8812au-5.13.6 for kernel 6.9.1-arch1-1 (x86_64)
Sun May 19 10:40:00 AM CEST 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.9.1-arch1-1/build M=/var/lib/dkms/rtl8812au/5.13.6/build  modules
make[1]: Entering directory '/usr/lib/modules/6.9.1-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/core/rtw_rf.o
[... 62 lines cut ...]
  469 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8811:6: note: declared here
 8811 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:488:9: error: too many arguments to function 'cfg80211_ch_switch_notify'
  488 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8795:6: note: declared here
 8795 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/hal/hal_btcoex_wifionly.o
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/lib/modules/6.9.1-arch1-1/build/Makefile:1919: /var/lib/dkms/rtl8812au/5.13.6/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.9.1-arch1-1/build'
make: *** [Makefile:2511: modules] Error 2`

nvzeppelin, 2024-05-19:
> @Finoderi @nvzeppelin 
> 
> I need both of you to move to the new version of this driver:
> 
> https://github.com/morrownr/8812au-20210820
> 
> Remove this driver using:
> 
> sudo sh remove-driver.sh
> 
> Then follow the installation steps for the new driver.
> 
> I have been busy and have failed to take this driver down but it will go down soon.

Fixed it thx. installed from aur so didnt see that there was another version

Finoderi, 2024-05-19:
OK, thanks.